scottmudge/OctoPrint-MeatPack #4: No settings or TX statistics

robdan7, 2021-01-14:
The title says it all. MeatPack shows up as an installed plugin, but there is no options page nor statistics visible in the web interface. I've installed the plugin twice and rebooted the Pi and printer. What could have gone wrong here? Anyone else with the same issue?

**System info:**
- Raspberry Pi Zero W Rev 1.1
- Prusa MK3S with MeatPack-firmware 3.9.3
- Octoprint version 1.5.2

scottmudge, 2021-01-14:
It sounds like the plugin isn't being loaded correctly. Perhaps there is a missing dependency. 

Can you send me your OctoPrint log file: `~/.octoprint/logs/octoprint.log` and `~/.octoprint/logs/serial.log`. You can also download them from the logging section of the OctoPrint settings. 

If there is any personal information, feel free to email me the logs - maill@scottmudge.com. I will delete them after review them.

robdan7, 2021-01-14:
Found the issue. Specifically a syntax error at [this line](https://github.com/scottmudge/OctoPrint-MeatPack/blob/ed581189090c00f39eed7f6133068c5662362dff/OctoPrint_MeatPack/__init__.py#L33). Turns out I'm only running python 3.5.3, even after a full update of Raspbian and python (what the heck?). Perhaps you should add minimum python version to the readme as a prerequisite _nudge nudge_ ;-)

closed 2021-01-14 by robdan7

scottmudge, 2021-01-14:
Thanks for letting me know, will do!

I can change that syntax as well, it was mostly for convenience.

robdan7, 2021-01-15:
Lol, no 3.5 doesn't work. Flashed a new OS and running Python 3.7 now. No more syntax errors!