scottmudge/OctoPrint-MeatPack #6: [Enhancement] Dynamic size display adjustment

FormerLurker, 2021-01-26:
Hi!  I have some code you might be able to use to improve the UI display a tiny bit.  Check out [this code](https://github.com/FormerLurker/ArcWelderPlugin/blob/bb71e8fe20cc65f0ab28c0c13685fde39f730320/octoprint_arc_welder/static/js/arc_welder.js#L115-L128) to adjust the units (b, kB, MB, GB, etc..) based on the supplied size in bytes and precision.  If it's not something you want/need, that's fine, but I thought you might be interested.

Anyway, I like your plugin, and am hoping your PR gets accepted.  If you can't get it into the prusa code, I think you should try the Marlin github.  Seems like a no brainer.

scottmudge, 2021-01-26:
I actually wanted something like this to scale the "KB" back to "MB" after 5 digits or so, so this is perfect. Thanks!

scottmudge, 2021-01-26:
Integrated, thanks again!

closed 2021-01-26 by scottmudge