scottmudge/OctoPrint-MeatPack #7: Error Loading MeatPack plugin

dsundt, 2021-01-26:
Hello - having issues trying to load the MeatPack Plugin into OctoPrint after installing via terminal approach, per the documentation provided.  I am using a Raspberry Pi Zero W, and my printer is a Prusa i3 MK3S.

I am able to successfully instal the plugin, but after rebooting OctoPrint I am receiving the following errors:

`2021-01-26 16:32:07,610 - octoprint.plugin.core - ERROR - Error loading plugin meatpack
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_MeatPack/__init__.py", line 10, in <module>
    from OctoPrint_MeatPack.packing_serial import PackingSerial
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_MeatPack/packing_serial.py", line 24
`
Thanks for your help.

scottmudge, 2021-01-26:
You'll need to update to Python v3.7. I believe the Zero might come with Python 3.5, which this is incompatible with.

scottmudge, 2021-01-26:
I am actually working on improving compatibility with Python 3.5, so in a few hours I'll have an update to fix this problem.

cp2004, 2021-01-26:
@scottmudge just to clarify, most OctoPrint installs are on Python 2. We are trying to push them to Python 3.7+ - the minimum for OctoPrint is 3.6, so don't worry about 3.5!

You'll need to adjust this line:

cp2004, 2021-01-26:
@scottmudge just to clarify, most OctoPrint installs are on Python 2. We are trying to push them to Python 3.7+ - the minimum for OctoPrint is 3.6, so don't worry about 3.5!

You'll need to adjust this line:
https://github.com/scottmudge/OctoPrint-MeatPack/blob/master/OctoPrint_MeatPack/__init__.py#L189

to match your python compatibility. I also recommend changing this bit in setup.py:
https://github.com/scottmudge/OctoPrint-MeatPack/blob/6ddd98e927407a0b1410ab678bbba740454f9392/setup.py#L55

to be:
```python
additional_setup_parameters = {"python_requires": ">=3.7, <4"}
```
so it refuses to install. Let me know if you have more questions. Feel free to hop into #dev-plugins on the [OctoPrint discord](https://discord.octoprint.org) if you want to chat with any of us, or just a ping on Github too.

cp2004, 2021-01-26:
Just one more thing, @dsundt follow the instructions here to upgrade to Python 3 from Python 2 (that you are running):
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

scottmudge, 2021-01-26:
Thanks for the insight. I've added support back to 2.7 (just minor syntax changes). But I do agree, people should be updating to newer versions of Python, both for performance and for security.

But the goal of this plugin is to improve data throughput on slower/older machines. Given that many of the RPi Zero installations that could benefit from this seem to have older versions of Python installed. Therefore I would like to expand compatibility as much as possible.

Compatibility for 2.7 was trivial for me, but I'm sure it will be a PITA for Octoprint as a whole. I will leave it up to OctoPrint to let users know they need to update, but for now I'd prefer to let as many people use it as possible.

cp2004, 2021-01-26:
Python version is not tied to Raspberry Pi model - all the Pi zeros can run OctoPi 0.18 or Python 3 just fine, as can all the way back on the RPi Model B.
Understand your reasons, it makes sense. OctoPrint is currently compatible with both, we are trying to get away from 2.7 as fast as possible so wherever I can I'm recommending people upgrade lol. Not something you have any need to do, that's just me wanting things to happen faster lol.

scottmudge, 2021-01-26:
Oh I know it's not tied intrinsically to the hardware, I just figured most people are going to be too lazy to upgrade the Python installation. 

As a compromise, I'll make a note in the README that it's highly recommended to update!

scottmudge, 2021-01-26:
@dsundt - Alright, OctoPrint-MeatPack v1.5.9 released just now should work with older Python versions. 

I haven't tested 2.7 personally, but the compatibility inspection tool I used didn't show any errors when using the 2.7 standards.

closed 2021-01-26 by scottmudge

dsundt, 2021-01-26:
@scottmudge Still having issues with 2.7, getting this error now:

`2021-01-26 23:19:00,510 - octoprint.plugin.core - ERROR - Error loading plugin meatpack
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_MeatPack/__init__.py", line 10, in <module>
    from OctoPrint_MeatPack.packing_serial import PackingSerial
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_MeatPack/packing_serial.py", line 24` 

@cp2004 - thanks for the script, but it looks like the Prusa OctoPi (version 0.15.0PE) image is not supported by your script.  My OctoPrint version is 1.5.3.  I'll have to look into the unsupported version instructions, or let me know if you have any other thoughts. Thanks

scottmudge, 2021-01-26:
Thanks for the info, I'll install python 2.7 and debug the issue. Give me a couple hours.

scottmudge, 2021-01-27:
@dsundt 

Okay I installed OctoPrint on 2.7 to debug this, and it works now. Had to remove all of the type hinting (makes me sad) to support 2.7, but it appears to load now.

Update to MeatPack 1.5.14.