scottmudge/OctoPrint-MeatPack #8: Can't connect to printer with Marlin built after 25/01/2021

daemuth, 2021-01-26:
Hi! I'm using the firmware from the [Marlin Firmware Service](https://marlin.crc.id.au/), built on 26/01/2021 on my Ender3v2 with BLTouch. (The service builtds the the bugfix-2.0 branch daily)

After installing MeatPack, connection attempts to my printer fail. This issue can be fixed by disabling MeatPack.
As far as I understand (based on the readme), this _should_ work, although I couldn't find the merged PR.

I suspect that this is due to the compression not being compatible with the firmware. Hopefully it's temporary, just wanted to let you know that others might be encountering the same issue.

Thanks for the plugin!

scottmudge, 2021-01-26:
Thanks for the report.

Indeed, it appears to be merged into that branch:

https://github.com/MarlinFirmware/Marlin/commit/3921369f98f39280800b1c9944677e9644278106

@ellensp - it looks like the preprocessor directive `ENABLED(MEATPACK)` needs to be true. Is this set to true by default in the CI/CD server for Marlin?

scottmudge, 2021-01-26:
Actually, that service may be third-party and unrelated to the Marlin repo.

I will add a feature request to ask them to add the `MEATPACK` directive as an option.

In the meanwhile, you may need to build a version of the firmware yourself.

daemuth, 2021-01-26:
Yes, AFAIK it is a third-party service. Thank you for submitting a feature request! Didn't know about the build requirement.

That explains and solves my issue and as such I'll be closing it with this comment.

closed 2021-01-26 by daemuth

leetwanker, 2021-01-26:
I'm having this same issue after building my own firmware. I enabled MeatPack in Configuration_adv.h:

`#define MEATPACK                // Support for MeatPack G-code compression (https://github.com/scottmudge/OctoPrint-MeatPack)
`

Do I need to do something else to get Octoprint to connect? 

Thanks!

scottmudge, 2021-01-27:
To be honest I'm not sure, I use Prusa's derivative. The Marlin port developer might have insight.

@ellensp are there any other build steps required to enable MP in the Marlin build?

I'll look into getting a simulator to test Marlin FW, but I'll need to install a Linux VM first.

scottmudge, 2021-01-27:
@leetwanker Yeah looking at the Marlin source code, it looks like it should work if `MEATPACK` is correctly defined and the firmware is cleanly rebuilt.

`octoprint.log` might have more information about why it's not working. It's possible the plugin may not be loading correctly.

scottmudge, 2021-01-27:
I discovered the problem, minor bug with the Marlin implementation.

I created a PR for the fix here (very simple), so if you want to build a functional version of Marlin yourself, look here: https://github.com/scottmudge/OctoPrint-MeatPack/issues/10#issuecomment-767987280