scottmudge/OctoPrint-MeatPack #10: can't connect to ender 3 v2 with meatpack plugin installed

swheettaos, 2021-01-27:
I'm running octoprint Version 1.5.3 on octopi Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.2.  When I install meatpack 1.5.14 I can't connect to the printer via USB anymore.  I have tried disabling the gcode compression and whitespace options.  I've also tried with a marlin firmware version from https://marlin.crc.id.au/firmware/Ender%203%20V2%20-%20v4.2.7%20Board%20-%20BLTouch/ .. If I use the previous version without meatpack or the new version with meatpack both have the same problem.  If the plugin is installed for meatpack I can't connect to the printer.  If I remove the plugin I can connect to the printer just fine.  I'm connecting to /ttyUSB0 @ 115200 with no problem.  RIght now I removed the plugin and using the EEPROM editor plugin for octoprint I can see the meatpack option is set to true (and was false on the previous version) .. I'll submit this and make sure I clean up the plugin (I had tried installing 1.5.9 earlier and it wasn't working.. I tried again and it installed 1.5.14 but I didn't do a fresh install.. I'll try that.)

swheettaos, 2021-01-27:
nope .. with fresh install of the plugin (removal and cleanup) still just can't connect .. it's timing out when trying to connect to the serial port.  If I go in and disable the plugin and restart octoprint it then works just fine.  What do you need from me to debug?

swheettaos, 2021-01-27:
I'm upgrading to python 3 to see if that fixes it.

scottmudge, 2021-01-27:
Marlin builds from marlin.crc.id.au do not have the `MEATPACK` mode enabled, so the firmware does not have it enabled.

You will need to enable this preprocessor directive yourself and compile the firmware. Unfortunately it is still a nascent feature, and CI/CD servers haven't yet been updated to enable this feature.

Without the feature enabled in the firmware configuration at build time, it will not work and you will not be able to connect.

swheettaos, 2021-01-27:
The latest nightly build has it enabled and I can see it in the firmware options set to true.  He just posted about it being enabled on on the firmware builds.. 

- Added support for Meatpack (2020-01-27 onwards)

scottmudge, 2021-01-27:
That's good it's been enabled as of the 27th, I was looking at the 25th build. What does your octoprint.log file look like?

swheettaos, 2021-01-27:
env.python.pip: 21.0
env.python.version: 3.7.3 

still not working ..

2021-01-27 02:51:20,566 - octoprint.startup - INFO - ******************************************************************************
2021-01-27 02:51:20,568 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-01-27 02:51:20,568 - octoprint.startup - INFO - ******************************************************************************
2021-01-27 02:51:21,035 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2021-01-27 02:51:21,199 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2021-01-27 02:51:21,199 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2021-01-27 02:51:21,199 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2021-01-27 02:51:21,209 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2021-01-27 02:51:21,237 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-01-27 02:51:22,223 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 9, in <module>
    import numpy as np
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import core
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/home/pi/oprint/bin/python"
  * The NumPy version is: "1.19.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

2021-01-27 02:51:22,303 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2021-01-27 02:51:22,323 - octoprint.plugin.core - INFO - Found 24 plugin(s) providing 24 mixin implementations, 56 hook handlers
2021-01-27 02:51:22,495 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
[... 134 lines cut ...]
2021-01-27 02:51:31,164 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 8.55ms
2021-01-27 02:51:31,243 - octoprint.server.util.sockjs - INFO - User wheetocto logged in on the socket from client ::ffff:10.0.0.60
2021-01-27 02:51:31,660 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.5.3', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '21.0', 'cores': 4, 'freq': 1500.0, 'ram': 3959308288, 'pi_model': 'Raspberry Pi 4 Model B Rev 1.2', 'octopi_version': '0.17.0'}
2021-01-27 02:51:31,988 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 250000
2021-01-27 02:51:31,998 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-01-27 02:51:31,999 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-27 02:51:32,989 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 15.67ms
[... cut ...]
2021-01-27 02:51:41,091 - octoprint.server.util.sockjs - INFO - User wheetocto logged in on the socket from client ::ffff:10.0.0.60
2021-01-27 02:51:43,058 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 13.79ms
2021-01-27 02:51:45,051 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 9.98ms
2021-01-27 02:51:47,005 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 13.95ms
2021-01-27 02:51:48,994 - tornado.access - WARNING - 409 GET /api/printer (::ffff:10.0.0.60) 7.50ms

swheettaos, 2021-01-27:
looks like the upgrade to python 3 broke the bed visualizer

scottmudge, 2021-01-27:
By the log it looks like the Marlin integration of MeatPack is not sending back the appropriate confirmation message to the OctoPrint plugin, so it is not enabling communication.

Another developer integrated MeatPack into Marlin, but I will look at their code to ensure it is working correctly.

swheettaos, 2021-01-27:
ok .. disabling plugin for now

scottmudge, 2021-01-27:
Unfortunately the Marlin code has been refactored/modified a lot from my original, but I will let them know they did not integrate it correctly.

I'm surprised they did not test it before merging.

scottmudge, 2021-01-27:
Okay I found the issue... they need to emit a space character after

`SERIAL_ECHOPGM(MeatPack_ProtocolVersion);` 

inside of 

`void MeatPack::report_state()`.

If you want, you can add the following line after line 207 in `meatpack.cpp`:

` SERIAL_ECHOPGM(MeatPack_ProtocolVersion);`

after that line, add: 

`SERIAL_CHAR(' ');`

And it should work.

scottmudge, 2021-01-27:
Okay I created a PR: https://github.com/MarlinFirmware/Marlin/pull/20896

I will update the readme with a notice.

FunkeCoder23, 2021-01-27:
I'm still running into this issue, brand new FW build with meatpack enabled, with your fix in report_state.
[octoprint.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5877633/octoprint.log)
[serial.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5877634/serial.log)

scottmudge, 2021-01-27:
Strange, it the firmware doesn't look like it's responding to the MP query commands, so the plugin assumes it's not properly connected.

Is that the entirety of the octoprint.log contents? In @swheettaos log, it was at least receiving the protocol version and whitespace removal flags. The issue there was there wasn't a space between "ON" and the protocol version, and the plugin is looking for that space.

But I don't see any query responses in your log. Which to me would indicate that it's not actually enabled in the firmware, for some reason.

FunkeCoder23, 2021-01-27:
```define MEATPACK``` is definitely uncommented. Here's the process I just did: booted new firmware, (re)enable meatpack plugin, reboot pi, here's the totality of the octoprint and serial logs
[serial.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5877696/serial.log)
[octoprint.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5877697/octoprint.log)

scottmudge, 2021-01-27:
Still looks like MeatPack isn't responding at all in the firmware. At the very least it should emit a protocol version. Can you attach a .elf file of the firmware you compiled? Should be in the output directory. The hex will be harder to disassemble, but the .elf should be a lot easier.

FunkeCoder23, 2021-01-27:
I had to change the extension for github to allow it:
[firmware.elf.txt](https://github.com/scottmudge/OctoPrint-MeatPack/files/5877712/firmware.elf.txt)

scottmudge, 2021-01-27:
Well the binary indicates it's definitely got MP methods linked into the serial parser, but for whatever reason the MP module is not receiving any of the incoming data. 

You should be seeing a line like

 `octoprint.plugins.meatpack - INFO - [Serial]: Config var [NoSpaces] synchronized (=disabled).` 

soon after the MeatPack module loads and emits its version number to the log.

Can you try adding:

`#define MEATPACK_LOOKUP_TABLE` 

to `Configuration_adv.h` and recompiling?

CRCinAU, 2021-01-27:
I run the site marlin.crc.id.au. I'm just experimenting with this now and came across this - and trying to get the hotfix going.

The custom builder has had the hotfix https://patch-diff.githubusercontent.com/raw/MarlinFirmware/Marlin/pull/20896 applied, so I've created one via the builder to test with.

The Terminal window logs:
```
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: PackingSerial<id=0x7f7839acf8, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
```

What specifics do you want to know?

CRCinAU, 2021-01-27:
Building again with `#define MEATPACK_LOOKUP_TABLE` enabled, and I can establish two way connectivity with the printer.

I do note that the `N0 M110 N0*125` seems to be repeated a couple of times - which seems strange?

```
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: PackingSerial<id=0x7f84077fd0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok N0 P15 B3
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Jan 27 2021 21:22:57) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:MEATPACK:1
```

ellensp, 2021-01-27:
after updateing to latest octoprint meatpack plugin I see this in octoprint.log I see 
2021-01-27 10:55:25,344 - octoprint.plugins.meatpack - INFO - [Serial]: Detected MeatPack protocol version V1
2021-01-27 10:55:25,356 - octoprint.plugins.meatpack - INFO - [Serial]: Config var [NoSpaces] synchronized (=disabled).
2021-01-27 10:55:25,386 - octoprint.plugins.meatpack - INFO - [Serial]: Config var [Enabled] synchronized (=enabled).
2021-01-27 10:55:25,397 - octoprint.plugins.meatpack - INFO - [Serial]: Config var [NoSpaces] synchronized (=disabled).
2021-01-27 10:55:25,409 - octoprint.plugins.meatpack - INFO - [Serial]: MeatPack configuration successfully synchronized and confirmed between host/device.
2021-01-27 10:55:25,423 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port

CRCinAU, 2021-01-27:
>  AttributeError: 'list' object has no attribute 'clear'

Which python version? I'm running OctoPi 0.18.0 RC1 which is 64 bit - but also Python 3.7...

scottmudge, 2021-01-27:
@ellensp Thanks, I'll remove "clear()" for Python 2.7 support (@CRCinAU - I'm guessing 2.7).

ellensp, 2021-01-27:
yes Python 2.7.16

scottmudge, 2021-01-27:
@ellensp I've pushed a compatibility fix for 2.7 (v1.5.17), can you confirm it works?

And @CRCinAU - the `N0 M110 N0*125` line is what I think is OctoPrint's way of "handshaking" with the firmware, so it does it a few times until an expected response is established (which is after MeatPack synchronizes the configuration state, which takes a second). Shouldn't hurt anything so long as it receives a response.

Also, has the build configuration changed for the available firmware builds to reflect `#define MEATPACK_LOOKUP_TABLE`? For the time being it looks like it might be a requirement for functionality.

CRCinAU, 2021-01-27:
> Also, has the build configuration changed for the available firmware builds to reflect `#define MEATPACK_LOOKUP_TABLE`? For the time being it looks like it might be a requirement for functionality.

Yeah - I've added both right now and rebuilt the 20210127 firmware (which the user will have to rename to a unique filename to flash - or wait until 20210128).

Also, the M84 easter egg just does a whole heap of single tone beeps.... Is that dependent on something else?

CRCinAU, 2021-01-27:
With the two defines - I got the following on my normal 20mm calibration cube print:
[image]

ellensp, 2021-01-27:
> I've pushed a compatibility fix for 2.7 (v1.5.17), can you confirm it works?
Yes this now get past this point. but getting errors about to enable MEATPACK_LOOKUP_TABLE and re try

scottmudge, 2021-01-27:
If the lookup table is the only functional mode, we may want to make that the default (without a define? or with the define pre-established in the configuration header) in the Marlin implementation.

ellensp, 2021-01-27:
with #define MEATPACK_LOOKUP_TABLE seems to be working
Comp. TX: 1.744 kB
Effective TX: 2.693 kB
Comp. Ratio: 0.648
Effective TX Rate: 64 Bytes/sec
Packing State: Enabled

CRCinAU, 2021-01-27:
> If the lookup table is the only functional mode, we may want to make that the default (without a define? or with the define pre-established in the configuration header) in the Marlin implementation.

Pretend I'm an idiot..... I'm not quite sure of the difference between using what I assume is a predefined lookup table for ASCII -> binary packing and not having it set via the MEATPACK_LOOKUP_TABLE... What operational difference is there?

scottmudge, 2021-01-27:
I had tested to see if a switch/case statement would perform faster than a lookup table in returning the packed character based on it's 4-bit representation.

However, when I looked at the disassembly in Ghidra, it was byte-for-byte identical. So the compiler seemed to optimize both implementations in the same way.

Unfortunately it looks like the switch/case method may not have been translated exactly correctly to Marlin (or perhaps it just wasn't updated on my end), so it did not work.

I always ended up using the table, as it allowed mutability for changing the library of packed characters, if ever I wanted to add that feature for compressing other types of data beyond 3D printing g-code.

ellensp, 2021-01-27:
scottmudge with  MEATPACK_LOOKUP_TABLE it has a static table 
```
  static const uint8_t meatPackLookupTable[16] = {
    '0', '1', '2', '3', '4', '5', '6', '7', '8', '9',
    '.', ' ', '\n', 'G', 'X',
    '\0' // Unused. 0b1111 indicates a literal character
  };

```
If you enable space removal... this is going to cuuse an issue
ie space isn't swapped with E so any encoded E's are going to be decoded as a space aren't they?

scottmudge, 2021-01-27:
@ellensp  - yes, in my Prusa code the table is mutable and the space character is swapped with an "E" (the replacement character define) when whitespace removal is enabled.

When I disassembled making it a static const and making it a mutable array, the disassembly was identical. There shouldn't be a performance benefit to making it static.

Also, do not make it PROGMEM, since it is referenced so frequently. Those extra cycles for accessing PROGMEM will add up.

CRCinAU, 2021-01-27:
Ok, this is all fixed in the new builds of firmware for 20210127 (or use 20210128 to avoid confusion) and MeatPack plugin v1.5.21 which is now available.

Thanks to @ellensp and @scottmudge for the collaboration over the last several hours to iron out every issue we could find and have something much more stable as a general release.

scottmudge, 2021-01-27:
Thanks @CRCinAU!

closed 2021-01-27 by scottmudge

swheettaos, 2021-01-27:
ty everyone.. it's working now!