scottmudge/OctoPrint-MeatPack #11: Checksum errors when using Remove All Whitespace ("G" Commands Only)

CRCinAU, 2021-01-27:
When this option is enabled, the checksum for commands fails - I believe because the spaces have been removed, the checksums need to be updated to reflect this.

Octoprint terminal log:
```
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Send: N1 G28 O*125
[...]
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: X:151.00 Y:115.00 Z:11.80 E:0.00 Count X:12080 Y:9200 Z:4720
Recv: ok N1 P15 B3
Send: N2 M113 S2*99
Recv: ok N2 P15 B3
Send: N3 G29 O*126
[...]
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.082 +0.075 +0.053 +0.058 +0.058
Recv:  1 +0.025 +0.048 +0.020 +0.013 +0.030
Recv:  2 +0.028 +0.025 +0.020 +0.005 +0.023
Recv:  3 +0.038 +0.035 +0.020 +0.015 +0.028
Recv:  4 +0.055 +0.055 +0.045 +0.035 +0.045
Recv: 
Recv: X:220.00 Y:220.00 Z:11.80 E:0.00 Count X:17600 Y:17600 Z:4720
Recv: ok N3 P15 B3
Send: N4 M500*34
[...]
Recv: ok N4 P15 B3
Send: N5 M420 S1*98
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00
Recv: ok N5 P15 B3
Changing monitoring state from "Starting" to "Printing"
Send: N6 M201 X500 Y500 Z100 E5000*12
[...]
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Recv: ok N2 P15 B4
Send: N6 M201 X500 Y500 Z100 E5000*12
Recv: Error:checksum mismatch, Last Line: 5
Recv: Resend: 6
Printer keeps requesting line 6 again and again, communication stuck
Changing monitoring state from "Printing" to "Error: Printer keeps requesting line 6 again and again, communication stuck"
Send: M112
Send: N7 M112*38
Send: N8 M104 T0 S0*41
Send: N9 M140 S0*108
Changing monitoring state from "Error: Printer keeps requesting line 6 again and again, communication stuck" to "Offline (Error: Printer keeps requesting line 6 again and again, communication stuck)"
Connection closed, closing down monitor
Closing down send loop
```

CRCinAU, 2021-01-27:
I should note that after this, I can't reconnect to the printer without powering it off and on again.... Maybe a problem with the checksum and the `N0 M110 N0*125` line on the connect attempt being seen as invalid?

scottmudge, 2021-01-27:
Let me check, I do recompute the checksum, but it looks like the "P" command is unexpected, and it is appending a checksum when it should not.

scottmudge, 2021-01-27:
Actually, let me check how marlin is computing checksums... it should be working in this case.

CRCinAU, 2021-01-27:
If it helps, the P is part of the ADVANCED_OK replies and received FROM the printer.

scottmudge, 2021-01-27:
I misread the error message, it looks like the "P" (line 5) was okay, but Line N6 had an issue. But it shouldn't have removed the whitespace from that line, as it was an "M" command. 

I'm about to push an update which explicitly restricts whitespace removal (and checksum updates) to "G" commands only, rather than just excluding "M" commands. Hopefully that corrects it.

scottmudge, 2021-01-27:
Alright let me know if this update helps:

https://github.com/scottmudge/OctoPrint-MeatPack/releases/tag/v1.5.18

It shouldn't touch the outgoing checksum unless it's a G-command, in which case N6 shouldn't fail the checksum check.

CRCinAU, 2021-01-27:
Oh no! :)

```
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: PackingSerial<id=0x7f4df40e80, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: TypeError: 'object of type 'NoneType' has no len()' @ comm.py:_readline:3831
Changing monitoring state from "Connecting" to "Offline (Error: TypeError: 'object of type 'NoneType' has no len()' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
```

scottmudge, 2021-01-27:
Oh geez, what's your log look like? Should have a better stack trace.

scottmudge, 2021-01-27:
Nevermind, I found the problem. Wrongly scoped return.

CRCinAU, 2021-01-27:
```
2021-01-27 12:02:18,103 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3831, in _readline
    ret = self._serial.readline()
  File "/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_MeatPack/packing_serial.py", line 285, in readline
    self._flush_buffer()
  File "/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_MeatPack/packing_serial.py", line 342, in _flush_buffer
    super(PackingSerial, self).write(self._process_line_bytes(line))
  File "/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_MeatPack/packing_serial.py", line 355, in _process_line_bytes
    return mp.pack_line(str_line)
  File "/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_MeatPack/meatpack.py", line 166, in pack_line
    line_len = len(line)
TypeError: object of type 'NoneType' has no len()
```

scottmudge, 2021-01-27:
Alright, pushed fix. Need a release, or can you pull the master zip?

CRCinAU, 2021-01-27:
Did a new install using: https://github.com/scottmudge/OctoPrint-MeatPack/archive/master.zip

Can talk to the printer ok - but after enabling the "Remove all Whitespace" option again, I get:
```
Changing monitoring state from "Operational" to "Starting"
[...]
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Changing monitoring state from "Starting" to "Printing"
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Recv: ok P15 B4
Send: N1 M201 X500 Y500 Z100 E5000*11
Recv: Error:checksum mismatch, Last Line: 0
Recv: Resend: 1
Printer keeps requesting line 1 again and again, communication stuck
Changing monitoring state from "Printing" to "Error: Printer keeps requesting line 1 again and again, communication stuck"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Printer keeps requesting line 1 again and again, communication stuck" to "Offline (Error: Printer keeps requesting line 1 again and again, communication stuck)"
Connection closed, closing down monitor
Closing down send loop
```

scottmudge, 2021-01-27:
Strange, it shouldn't even be touching the `M201` line, or affecting the checksum its sending out with it.

I might create a debug branch that dumps the outgoing data to the log so I can verify this is the case.

scottmudge, 2021-01-27:
Alright, thanks for helping me debug this.

I've pushed a branch which will output all outgoing/packed data to octoprint.log.

You can install it with `pip install git+https://github.com/scottmudge/OctoPrint-MeatPack.git@debug` .

If you wouldn't mind attaching the octoprint.log file after doing this, it will let me know if/how the checksum is being affected.

CRCinAU, 2021-01-27:
If you happen to use Discord, I'm on the Marlin discord channels if you wanna yell out in #support? 

Might be easier than going via the issue for further debugging?

scottmudge, 2021-01-27:
Alright, I'm in the support channel now. Your username the same?

CRCinAU, 2021-01-27:
Ok, so this is fixed with the combination of build 20210128 or later and plugin version 1.5.21 or higher.

closed 2021-01-27 by CRCinAU