scottmudge/OctoPrint-MeatPack #12: OctoPrint unable to connect to printer when plug-in is enabled

Taomyn, 2021-01-28:
I have an Ender 3 Pro with SKR BTT Mini E3 v1.2, OctoPi v0.18 (RPi-4) fully up to date, MeatPack plug-in v 1.5.21. My fork of Marlin can be found here https://github.com/Taomyn/Marlin

Compiled latest bugfix as I found tonight including enabling MeatPack.

When enabled OctoPrint will not connect to the printer, disable it and OctoPrint is working fine. Some log data below, let me know if I can provide anything else.

Excerpt from OctoPrint log:

```
2021-01-28 20:03:25,462 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2021-01-28 20:03:25,482 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 2 port/baudrate candidates: /dev/ttyACM0@250000, /dev/ttyS0@250000
2021-01-28 20:03:25,482 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 250000
2021-01-28 20:03:25,483 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
2021-01-28 20:03:25,484 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
2021-01-28 20:03:25,507 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-01-28 20:03:25,511 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-28 20:03:27,721 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-01-28 20:03:27,726 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-28 20:03:29,753 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-01-28 20:03:29,757 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-01-28 20:03:32,753 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyS0, baudrate 250000
2021-01-28 20:03:32,763 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port /dev/ttyS0, baudrate 250000 from hook meatpack: SerialException: 'Could not configure port: (5, 'Input/output error')' @ comm.py:_open_serial:3670
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialposix.py", line 398, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
termios.error: (5, 'Input/output error')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3670, in _open_serial
    settings().getFloat(["serial", "timeout", "connection"]),
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_MeatPack/__init__.py", line 52, in serial_factory_hook
    self._serial_obj.port = str(port)
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialutil.py", line 276, in port
    self.open()
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
2021-01-28 20:03:32,802 - octoprint.util.comm - INFO - Serial detection: Could not open port /dev/ttyS0, baudrate 250000, skipping
2021-01-28 20:03:32,802 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2021-01-28 20:03:32,804 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2021-01-28 20:03:32,811 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-error' with value Error: No more candidates to test, and no working port/baudrate combination detected. (mydomain)
2021-01-28 20:03:33,180 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-error' with value Error: No more candidates to test, and no working port/baudrate combination detected. (mydomain)
```

Serial.log - failing
```
2021-01-28 20:02:47,456 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2021-01-28 20:03:25,445 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2021-01-28 20:08:08,102 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2021-01-28 20:10:42,155 - Enabling serial logging
2021-01-28 20:10:52,495 - Changing monitoring state from "Offline" to "Detecting serial connection"
2021-01-28 20:10:52,531 - Performing autodetection with 2 port/baudrate candidates: /dev/ttyACM0@250000, /dev/ttyS0@250000
2021-01-28 20:10:52,531 - Trying port /dev/ttyACM0, baudrate 250000
2021-01-28 20:10:52,549 - Handshake attempt #1 with timeout 2.0s
2021-01-28 20:10:52,552 - Connected to: PackingSerial<id=0xa8c07a70, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2021-01-28 20:10:52,554 - Send: N0 M110 N0*125
2021-01-28 20:10:53,141 - Recv: Not SD printing
2021-01-28 20:10:54,132 - Recv:  T:21.87 /0.00 B:34.17 /0.00 @:0 B@:0
2021-01-28 20:10:54,142 - Recv: Not SD printing
2021-01-28 20:10:55,143 - Recv: Not SD printing
2021-01-28 20:10:55,144 - Handshake attempt #2 with timeout 2.0s
2021-01-28 20:10:55,157 - Send: N0 M110 N0*125
2021-01-28 20:10:56,131 - Recv:  T:20.94 /0.00 B:34.17 /0.00 @:0 B@:0
2021-01-28 20:10:56,142 - Recv: Not SD printing
2021-01-28 20:10:57,142 - Recv: Not SD printing
2021-01-28 20:10:58,131 - Recv:  T:21.25 /0.00 B:34.14 /0.00 @:0 B@:0
2021-01-28 20:10:58,139 - Handshake attempt #3 with timeout 2.0s
2021-01-28 20:10:58,147 - Send: N0 M110 N0*125
2021-01-28 20:10:58,150 - Recv: Not SD printing
2021-01-28 20:10:59,143 - Recv: Not SD printing
2021-01-28 20:11:00,131 - Recv:  T:21.41 /0.00 B:34.17 /0.00 @:0 B@:0
2021-01-28 20:11:00,143 - Recv: Not SD printing
2021-01-28 20:11:00,145 - Trying port /dev/ttyS0, baudrate 250000
2021-01-28 20:11:00,156 - Unexpected error while connecting to serial port /dev/ttyS0, baudrate 250000 from hook meatpack: SerialException: 'Could not configure port: (5, 'Input/output error')' @ comm.py:_open_serial:3670
2021-01-28 20:11:00,192 - Could not open port /dev/ttyS0, baudrate 250000, skipping
2021-01-28 20:11:00,193 - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2021-01-28 20:11:00,194 - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2021-01-28 20:11:00,195 - Connection closed, closing down monitor
2021-01-28 20:11:07,075 - Disabling serial logging
```

Serial.log - working
```
2021-01-28 20:22:37,140 - Changing monitoring state from "Offline" to "Detecting serial connection"
2021-01-28 20:22:37,760 - Performing autodetection with 2 port/baudrate candidates: /dev/ttyACM0@250000, /dev/ttyS0@250000
2021-01-28 20:22:37,761 - Trying port /dev/ttyACM0, baudrate 250000
2021-01-28 20:22:37,762 - Connecting to port /dev/ttyACM0, baudrate 250000
2021-01-28 20:22:37,868 - Handshake attempt #1 with timeout 2.0s
2021-01-28 20:22:37,893 - Connected to: Serial<id=0xa8b76910, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2021-01-28 20:22:37,895 - Send: N0 M110 N0*125
2021-01-28 20:22:37,931 - Recv: ok WNÂ…0g P31 B31
2021-01-28 20:22:37,978 - Changing monitoring state from "Detecting serial connection" to "Operational"
2021-01-28 20:22:38,036 - Send: N0 M110 N0*125
2021-01-28 20:22:38,116 - Recv: ok WNÂ…0g P31 B31
2021-01-28 20:22:38,118 - Send: N1 M115*39
2021-01-28 20:22:38,155 - Recv: FIRMWARE_NAME:Marlin v2.0.x (Jan 28 2021 19:53:47) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1 UUID:
```

scottmudge, 2021-01-28:
Looks like it's trying to access a port that isn't available (`/dev/ttyS0`), and the exception isn't being caught in the python module.

I have implemented a wrapper around the standard Serial object. I'm guessing OctoPrint typically just discards these exceptions if the port isn't available.

You can try disabling or removing the `/dev/ttyS0` port, since it looks like you are using `/dev/ttyACM0`. In the meanwhile, I will try/except the exception and discard it, since it's only telling us the port can't be opened.

scottmudge, 2021-01-28:
Actually, you should really specify the correct port. The failing MeatPack log shows a different port being used than the one logged in the "working" log. Does MeatPack fail with `/dev/ttyACM0`?

Taomyn, 2021-01-28:
> Actually, you should really specify the correct port. The failing MeatPack log shows a different port being used than the one logged in the "working" log. Does MeatPack fail with `/dev/ttyACM0`?

Nope just sits there "Connecting"......

[image]

scottmudge, 2021-01-28:
Alright you can try installing this branch which catches that exception. It might just be trying ttyS0 and then it can't recover.

You can install by loading your OctoPrint virtualenv and then running:

`pip install git+https://github.com/scottmudge/OctoPrint-MeatPack.git@bugfix_serialconnfail`.

If that doesn't work, I'll need the rest of the octoprint.log file, since the excerpt is missing info about the `ttyACM0` connection attempt.

Taomyn, 2021-01-28:
> If that doesn't work, I'll need the rest of the octoprint.log file, since the excerpt is missing info about the `ttyACM0` connection attempt.

No joy I'm afraid:
```
2021-01-28 20:46:59,600 - octoprint.events.fire - DEBUG - Firing event: SettingsUpdated (Payload: {'config_hash': '906e544d04923db7633c7ef4ffed074a', 'effective_hash': '58fbf453d6222da9ad45fc7d733c5a7d'})
2021-01-28 20:46:59,600 - octoprint.events - DEBUG - Sending action to <function Server.run.<locals>.on_settings_update at 0xaf947b70>
2021-01-28 20:46:59,601 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xa98505f0>>
2021-01-28 20:46:59,602 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-01-28 20:46:59,603 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-01-28 20:46:59,604 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-01-28 20:46:59,605 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-01-28 20:46:59,607 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-01-28 20:46:59,608 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-01-28 20:46:59,609 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-01-28 20:46:59,610 - octoprint.plugin - DEBUG - Calling on_event on DisplayLayerProgress
2021-01-28 20:47:01,637 - octoprint.plugin - DEBUG - Calling on_event on DryRun
2021-01-28 20:47:01,637 - octoprint.plugin - DEBUG - Calling on_event on FriendlyBeeper
2021-01-28 20:47:01,637 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2021-01-28 20:47:01,638 - octoprint.plugin - DEBUG - Calling on_event on backupscheduler
2021-01-28 20:47:01,639 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2021-01-28 20:47:01,639 - octoprint.plugin - DEBUG - Calling on_event on dashboard
2021-01-28 20:47:01,639 - octoprint.plugin - DEBUG - Calling on_event on eeprom_marlin
2021-01-28 20:47:01,640 - octoprint.plugin - DEBUG - Calling on_event on m73progress
2021-01-28 20:47:01,640 - octoprint.plugin - DEBUG - Calling on_event on printoid
2021-01-28 20:47:01,640 - octoprint.plugin - DEBUG - Calling on_event on tasmota
2021-01-28 20:47:05,899 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 6.87ms
2021-01-28 20:47:05,910 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 7.51ms
2021-01-28 20:47:06,111 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 8.15ms
2021-01-28 20:47:06,123 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 8.75ms
2021-01-28 20:47:06,297 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 7.39ms
2021-01-28 20:47:06,521 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 5.32ms
2021-01-28 20:47:06,702 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.1.1) 4.90ms
2021-01-28 20:47:08,575 - octoprint.events.fire - DEBUG - Firing event: Connecting (Payload: None)
2021-01-28 20:47:08,576 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xa98505f0>>
2021-01-28 20:47:08,577 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-01-28 20:47:08,578 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-01-28 20:47:08,579 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-01-28 20:47:08,579 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-01-28 20:47:08,580 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-01-28 20:47:08,580 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-01-28 20:47:08,581 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-01-28 20:47:08,581 - octoprint.plugin - DEBUG - Calling on_event on DisplayLayerProgress
2021-01-28 20:47:08,581 - octoprint.plugin - DEBUG - Calling on_event on DryRun
2021-01-28 20:47:08,581 - octoprint.plugin - DEBUG - Calling on_event on FriendlyBeeper
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on backupscheduler
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on dashboard
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on eeprom_marlin
2021-01-28 20:47:08,582 - octoprint.plugin - DEBUG - Calling on_event on m73progress
2021-01-28 20:47:08,583 - octoprint.plugin - DEBUG - Calling on_event on printoid
2021-01-28 20:47:08,583 - octoprint.plugin - DEBUG - Calling on_event on tasmota
2021-01-28 20:47:08,622 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2021-01-28 20:47:08,626 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
2021-01-28 20:47:08,632 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {'state_id': 'OPEN_SERIAL', 'state_string': 'Opening serial connection'})
2021-01-28 20:47:08,634 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
[... cut ...]
2021-01-28 20:48:14,407 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2021-01-28 20:48:14,408 - octoprint.plugin - DEBUG - Calling on_event on dashboard
2021-01-28 20:48:14,434 - octoprint.plugin - DEBUG - Calling on_event on eeprom_marlin
2021-01-28 20:48:14,446 - octoprint.plugin - DEBUG - Calling on_event on m73progress
2021-01-28 20:48:14,472 - octoprint.plugin - DEBUG - Calling on_event on printoid
2021-01-28 20:48:14,498 - octoprint.plugin - DEBUG - Calling on_event on tasmota

```

scottmudge, 2021-01-28:
Looks like it's still trying to read from `ttyS0` in that excerpt. The PackingSerial() class doesn't decide which port to open. OctoPrint tells it what port to open.

I'm guessing there's still more to the log. Can you upload the whole thing to pastebin or attach it?

Also, I see your branch has a lot of configuration differences from the default, but does the build from here work?:

https://marlin.crc.id.au/

If there's a compatible build, anyway. I've confirmed it works the other night with the guy who runs the site and creates those builds, so it would narrow down if it's something in your build or not.

scottmudge, 2021-01-28:
And a final note, when Marlin merged my file they did some refactoring which may not have been tested. The builds that were tested were built from a patch without this refactoring.

If you're able to use the crc.id.au builds and they work, try replacing "meatpack.cpp" on your branch with the one I've attached here, and do a clean rebuild.

[meatpack_cpp.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5889714/meatpack_cpp.zip)

Taomyn, 2021-01-28:
> Looks like it's still trying to read from `ttyS0` in that excerpt. The PackingSerial() class doesn't decide which port to open. OctoPrint tells it what port to open.
> 
> I'm guessing there's still more to the log. Can you upload the whole thing to pastebin or attach it?
> 

Nope that was everything for the time I was testing

> Also, I see your branch has a lot of configuration differences from the default, but does the build from here work?:
>
> https://marlin.crc.id.au/
>
> If there's a compatible build, anyway. I've confirmed [...]

Unfortunately I'm not going to be able to try that until the weekend, probably Sunday, and then it will take time as I'd need to be 100% it was going to work with my hardware set-up - it's hardly stock but not anything non-standard e.g. it's a genuine BL-Touch not a clone.

Taomyn, 2021-01-28:
> And a final note, when Marlin merged my file they did some refactoring which may not have been tested. The builds that were tested were built from a patch without this refactoring.
> 
> If you're able to use the crc.id.au builds and they work (or if you can't and it still doesn't work on your branch), try replacing "meatpack.cpp" on your branch with the one I've attached here, and do a clean rebuild.
> 
> [meatpack_cpp.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5889714/meatpack_cpp.zip)

Sure, I can try that tomorrow and let you know.

scottmudge, 2021-01-28:
No problem, then for the time being try replacing meatpack.cpp with the file I linked above, and then rebuild.

And if you wouldn't mind, after it's done compiling, rename the .elf file produced to .elf.txt and attach it to the response, so I can check the disassembly.

Thanks for helping in the debug process!

Taomyn, 2021-01-29:
> No problem, then for the time being try replacing meatpack.cpp with the file I linked above, and then rebuild.
> 
It would not compile

[image]

I did have issues trying to compile just the latest bugfix changes from today, but after I reverted it was ok again and then I copied over your file and the above is what I got.

scottmudge, 2021-01-29:
Indeed, the toggle command was removed in the header. Here's a fix, removed that case statement.
[meatpack_cor.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5895363/meatpack_cor.zip)

Taomyn, 2021-01-29:
> Indeed, the toggle command was removed in the header. Here's a fix, removed that case statement.
> [meatpack_cor.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5895363/meatpack_cor.zip)

[image]

scottmudge, 2021-01-29:
Geez they refactored a lot, that additional serial iface index might have something to do with why it doesn't work?

Here, I made a branch of the commit that was tested and confirmed working on my fork. See if you can merge your fork with it and build. It's pretty much the same as bugfix-2.0.x, just without the refactoring they did to MeatPack after they merged my PR.

https://github.com/scottmudge/Marlin/tree/bugfix-2.0.x_meatpack_test

Taomyn, 2021-01-29:
> Geez they refactored a lot, that additional serial iface index might have something to do with why it doesn't work?
>
> Here, I made a branch of the commit that was tested and confirmed working on my [...]

Sorry, I have no clue how to merge my branch with yours, I've only been using Github for my own fork for the last 6 months - I use Github Desktop on Windows, and even on the site I can't figure it out and none of the drop downs ever shows your fork, e.g.

[image]

It's also annoying that even after months hitting the "Compare" button any fork including the main Marlin one just brings up this

[image]

I reported it and they still haven't done anything about it.

scottmudge, 2021-01-29:
No problem, I can do it for you. Just give me a moment.

scottmudge, 2021-01-29:
Alright, so create a directory to store the repo, open a terminal inside of it, then run the following:

```
git clone https://github.com/scottmudge/Marlin.git
cd Marlin
git checkout Taomyn_Ender3Pro_Branch_MPMerge
```

That should get you a nice merged branch, with all your changes from your fork.

Taomyn, 2021-01-29:
> That should get you a nice merged branch, with all your changes from your fork.

Thanks, that was successfully compiled - I just need to wait about an hour for a test print to finish then I can flash it. Here's the .elf file in the meantime.

[firmware.elf.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5895754/firmware.elf.zip)

Di3mex, 2021-01-29:
I am having the same issue with my printer not connecting. I first merged the bugfix-2.0.0 branch from Marlin. It didn't connect.

Then I tried to automatically merge from the branch: bugfix-2.0.x_meatpack_test

Didn't compile.

Then I manually copy and pasted the meatpack files and got compile errors:

> In file included from Marlin\src\gcode\queue.cpp:52:
> Marlin\src\gcode\../feature/meatpack.h:117:55: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
>   117 |   static uint8_t unpacked_char(register const uint8_t in);
>       |                                                       ^~
> Marlin\src\gcode\queue.cpp: In static member function 'static void GCodeQueue::get_serial_commands()':
> Marlin\src\gcode\queue.cpp:482:46: error: no matching function for call to 'MeatPack::handle_rx_char(uint8_t, uint8_t&)'
>   482 |         meatpack.handle_rx_char(uint8_t(c), p);
>       |                                              ^
> In file included from Marlin\src\gcode\queue.cpp:52:
> Marlin\src\gcode\../feature/meatpack.h:105:15: note: candidate: 'static void MeatPack::handle_rx_char(uint8_t)'
>   105 |   static void handle_rx_char(const uint8_t c);
>       |               ^~~~~~~~~~~~~~
> Marlin\src\gcode\../feature/meatpack.h:105:15: note:   candidate expects 1 argument, 2 provided
> *** [.pio\build\LPC1768\src\src\gcode\queue.cpp.o] Error 1

I then copy and pasted the files from your initial commit (meatpack.h, .c and queue)

It compiled but still doesn't connect.

I am now reverting back so I can continue printing.

scottmudge, 2021-01-29:
> I then copy and pasted the files from your initial commit (meatpack.h, .c and queue)

@Di3mex Which commit did you copy from? Looking at your fork, it looks like it still has the refactored and potentially broken code from Marlin.

I need to bring it up with Marlin. It doesn't seem like they effectively tested all of the code they refactored when they pulled my changes, which is unfortunate.

I will prepare you  a branch from your fork with the tested, working revision.

scottmudge, 2021-01-29:
@Di3mex 

Try this branch, I copied your config:

https://github.com/scottmudge/Marlin/tree/Dimex_Branch_testing

```
git clone https://github.com/scottmudge/Marlin.git
cd Marlin
git checkout Dimex_Branch_testing
```

Taomyn, 2021-01-29:
@scottmudge I'm afraid the version compiled from your code is doing the same thing - when I set it to the port manually it just sits there forever connecting.

scottmudge, 2021-01-29:
Okay, well at least that's been narrowed down. If the only difference is the `Configuration.h` and `Configuration_adv.h` changes, then something in there must be causing a problem.

Di3mex, 2021-01-29:
> @Di3mex
> 
> Try this branch, I copied your config:
> 
> https://github.com/scottmudge/Marlin/tree/Dimex_Branch_testing
> 
> ```
> git clone https://github.com/scottmudge/Marlin.git
> cd Marlin
> git checkout Dimex_Branch_testing
> ```

Thanks, but also doesn't connect.

scottmudge, 2021-01-29:
I think I may have found a potential cause. The `#define TX_BUFFER_SIZE 0` set in your `Configuration_adv.h` file may need to be bumped up to `32`, as it is on the primary repo.

scottmudge, 2021-01-29:
Alright so there is some configuration difference between the builds produced on https://marlin.crc.id.au/ and those you're producing.

Can you produce a standard set of Configuration.h and Configuration_adv.h for your respective printer(s), or detail how yours is changed from the default? The existing one on the repo just looks like a template.

Taomyn, 2021-01-29:
> Alright so there is some configuration difference between the builds produced on https://marlin.crc.id.au/ and those you're producing.
>
> Can you produce a standard set of Configuration.h and [...]

Got a bit confused by your now deleted message as I was sure mine was 32
[image]

My readme.md lists all the changes I make to a default Marlin config files, I have not used any of the vendor files.

Di3mex, 2021-01-29:
@scottmudge The config in my repo is not the default config. Not much changed. Just ABL and the options to enable Host command support (in config.adv), Lin advance, 32 microsteps, bigger buffers and stuff to "improve" the serial connection via USB.

scottmudge, 2021-01-29:
@Taomyn yes I was looking at the wrong side of my diff view, the other (Marlin default) was set to 0. Does setting it to 0 have any effect though?

And on that note, @Di3mex , did you do the same modification to `TX_BUFFER_SIZE`? 

That's the only difference I can think of that would have a potential cause in the Firmware.

The only other potential cause would be on the host side. @Di3mex, can you link/attach your complete octoprint.log file? Perhaps OctoPrint + MeatPack doesn't like it when there's a separate, unused serial interface (e.g., `/dev/ttyS0`).

CRCinAU, 2021-01-30:
Ok - just to fix some confusion here - Meatpack won't work with USB Emulated serial - ie /dev/ttyACM0 - that is because it isn't real serial. There's no such thing as a baud rate in USB Emulated Serial - hence your max speeds are about 12Mbit - not 115200.

As such, it doesn't really make sense to use MeatPack in this scenario - as your printer can't print at a 12Mbit data rate.

As such, Marlin will only allow MeatPack on real serial ports (0-7) - whereas USB is serial port `-1`.

scottmudge, 2021-01-30:
Thank you for clarifying, should have noticed it was `ACM0`, not `AMA0`. 

On that note, I should pass-through control in the plugin w/o MP enabled if it is USB emulated. @CRCinAU - Is this type of serial device always enumerated with the `ttyACM*` device ID?

CRCinAU, 2021-01-30:
> Is this type of serial device always enumerated with the `ttyACM*` device ID?

I believe so.

hooksea, 2021-01-31:
Hi I just started to use this plugin with my Ender 3 pro using the firmware compiled on https://marlin.crc.id.au/ and I am experiencing this same issue. When the plugin is enabled, the octoprint doesn't connect over the USB serial connection. If the port is manually changed to /dev/ttyACM0 it sits at connecting and then eventually fails. Has this been resolved at all. Thanks for the great work!

ixendrax, 2021-01-31:
Hi having the same issue with Ender 3 V2 and standard E3 V2 BLTouch firmware from https://marlin.crc.id.au/ 
Using /dev/ttyUSB0

CRCinAU, 2021-01-31:
There's a lot of issues right now that Meatpack is falling victim to - and the majority are do to with serial rewrite done in the last few days...

Monitor these:
https://github.com/MarlinFirmware/Marlin/pull/20932

Once that gets merged, it should hopefully fix a lot of breakage that is currently going on...

X-Ryl669, 2021-01-31:
Please test with latest bugfix, it should be resolved. If so, please close the issue. Thanks!

CRCinAU, 2021-01-31:
Updated binaries are now on https://marlin.crc.id.au/ - look for 20210201 as the datestamp.

I'm not sure if it will still work on /dev/ttyACM* style 'serial' - but worth testing...

Taomyn, 2021-02-01:
> Please test with latest bugfix, it should be resolved. If so, please close the issue. Thanks!

Would love to but

```
Compiling .pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o
In file included from c:\users\fernando\documents\github\marlin\marlin\src\core\serial_hook.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/MarlinSerial.h:29,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:49,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:81:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(char c, int base = 0)              { print((long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:82:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(unsigned char c, int base = 0)     { print((unsigned long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:83:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(int c, int base = DEC)             { print((long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:84:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(unsigned int c, int base = DEC)    { print((unsigned long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:98:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void println(const char s[])                  { print(s); println(); }
   ^~~~~~~~~
   _NOINLINE
compilation terminated due to -fmax-errors=5.
In file included from c:\users\fernando\documents\github\marlin\marlin\src\core\serial_hook.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/MarlinSerial.h:29,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:49,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:81:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(char c, int base = 0)              { print((long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:82:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(unsigned char c, int base = 0)     { print((unsigned long)c, base); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:83:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(int c, int base = DEC)             { print((long)c, base); }
   ^~~~~~~~~
   _NOINLINE
In file included from c:\users\fernando\documents\github\marlin\marlin\src\core\serial_hook.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/MarlinSerial.h:29,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:49,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:81:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(char c, int base = 0)              { print((long)c, base); }
   ^~~~~~~~~
   _NOINLINE
In file included from c:\users\fernando\documents\github\marlin\marlin\src\core\serial_hook.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/MarlinSerial.h:29,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:49,
                 from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:81:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(char c, int base = 0)              { print((long)c, base); }
   ^~~~~~~~~
[... 136 lines cut ...]
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:84:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void print(unsigned int c, int base = DEC)    { print((unsigned long)c, base); }
   ^~~~~~~~~
   _NOINLINE
compilation terminated due to -fmax-errors=5.
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:98:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
   NO_INLINE void println(const char s[])                  { print(s); println(); }
   ^~~~~~~~~
   _NOINLINE
c:\users\fernando\documents\github\marlin\marlin\src\core\serial_base.h:98:3: error: 'NO_INLINE' does not name a type; did you mean '_NOINLINE'?
[... cut ...]
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_512K\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
================================================================================== [FAILED] Took 12.84 seconds ==================================================================================
```

CRCinAU, 2021-02-01:
This is a problem with your environment - not the updated bugfix.

Also, on a STM32F103RC_btt_512K, you won't be able to use MeatPack over the usb cdc connection - but given it has a max speed of 12Mbit, I'm not sure that matters.

X-Ryl669, 2021-02-01:
NO_INLINE is defined in [macros.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/macros.h). It's very likely you don't use bugfix, or you've missed a change in your fork?

X-Ryl669, 2021-02-01:
BTW, interesting project!

CRCinAU, 2021-02-01:
@X-Ryl669 - If you feel like it - MeatPack doesn't currently work on USB CDC (SERIAL_PORT -1) in Marlin.

We think that it probably should, even though it really doesn't add any benefit to transfer speeds over the USB CDC connection - but more for compatibility and staying consistent in available features between different types of "serial" connections...

If you happen to know how to fix this, it'd be a lovely thing to have... :)

X-Ryl669, 2021-02-01:
What is the issue ?

Taomyn, 2021-02-01:
> This is a problem with your environment - not the updated bugfix.
> 

@X-Ryl669 I'll revert back to my build from yesterday, disable Meatpack as I'm done with it and try applying the latest again.

CRCinAU, 2021-02-01:
I'm not exactly sure - only that it works on UART0-7 - but not -1 - being the USB port.

Taomyn, 2021-02-01:
Ok, so after reverting to yesterday's build and disabling Meatpack the compile worked again. I re-merged bugfix into my branch and it then compiled. Out of curiosity I re-enabled Meatpack and it still compile correctly - I have no clue why it was failing earlier but seems to be fixed now.

I still don't understand the issue with the USB on my board - as I understand it just like my USB serial console cable it's just a USB-to-Serial-bridge and so the operating system is given a standard serial port, COM3 albeit an emulated one, and anything using it treats it as such including the speed. It will never do 12Mbit, only the baud rate the application uses when connecting to the COM port. I'm happy to be shown why I am wrong.

X-Ryl669, 2021-02-01:
With CDC serial (the "USB serial comm port emulation"), the UART speed is completely ignored, since UART's data does not flow with 2 wires and strict clocking, it's captured and send as bytes in USB packets (at native USB speed, 12MBits 480 MBits whatever). So you were probably doing < 12Mbits serial without even knowing it. 

Can you describe the issue a bit more ? 
1. Without enabling Meatpack, do you get data from that USB's serial link in a terminal/octoprint ?
2. If yes, then Meatpack is probably doing something unexpected on the serial link (like, for example, trying to read 2 bytes while there's only one available. On a usual serial port, it'll just work with a single byte returned, but for USB, it's waiting for the 2 bytes to come, so you don't even get the former byte ?)
3. If no, then it's an issue with Marlin, and I need to figure out why it happens.

Taomyn, 2021-02-01:
> 1. Without enabling Meatpack, do you get data from that USB's serial link in a terminal/octoprint ?
Yes

Not sure how to answer the other questions, but thanks for explaining the CDC serial

CRCinAU, 2021-02-01:
@X-Ryl669 If you get on Discord, have a chat to "Dust" - he knows more about it than I do and can probably help you a long way with it...

X-Ryl669, 2021-02-01:
Is it related: https://github.com/MarlinFirmware/Marlin/issues/20367 ?

Snargaloo, 2021-02-02:
I am eager to try this MeatPack feature.  I have compiled the latest Marlin 2.0.x-bugfix from this morning.  I use git stash to apply my changes to the latest rebase.  I will attach the patch file from my git stash.  I have latest OctoPrint running on an Rpi4 with a Python3 only environment.  I have the RPi4 Bluetooth device disabled making the first PL011 (UART0) the primary UART.  I connect to my printer using /dev/ttyAMA0.  When MeatPack is enabled, OctoPrint will not connect.  Disabling MeatPack allows OctoPrint to connect again.
[serial_with_meatpack.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5913913/serial_with_meatpack.log)
[serial_without_meatpack.log](https://github.com/scottmudge/OctoPrint-MeatPack/files/5913918/serial_without_meatpack.log)
[marlin_2_0_x_bugfix_changes.zip](https://github.com/scottmudge/OctoPrint-MeatPack/files/5913930/marlin_2_0_x_bugfix_changes.zip)

psyvision, 2021-02-02:
I was also looking at using this, but reading above not sure if I'd see a benefit.  

Anyway, with **ttyACM0** in use on Marlin bugfix-2.0 forked this morning I cannot connect with OctoPrint.  If I disable the OctoPrint plugin it works fine.  Marlin is configured to use `SERIAL_PORT 0`.

Can I provide any logs or try additional code to help?

Foxies-CSTL, 2021-02-04:
I wanted to try your plugin (v15.5.21) and I also encounter connection problems with Octoprint when it is activated.
My connection is 500k in UART0 (Serial0/PL011 aka ttyAMA0) on the raspberry and UART2 (Tx/Rx from the Wifi port) on my Delta printer.
GPIO Port:
```
"Connecting to port / dev / serial0, baudrate 500000
Handshake attempt # 1 with timeout 2.0s
Connected to: Serial <id = 0x66cf57d0, open = True> (port = '/ dev / serial0', baudrate = 500000, bytesize = 8, parity = 'N', stopbits = 1, timeout = 2.0, xonxoff = False, rtscts = False, dsrdtr = False), starting monitor "
```
USB port:
```
"Connecting to port / dev / ttyUSB0, baudrate 500000
Handshake attempt # 1 with timeout 2.0s
Connected to: Serial <id = 0x66d25d50, open = True> (port = '/ dev / ttyUSB0', baudrate = 500000, bytesize = 8, parity = 'N', stopbits = 1, timeout = 2.0, xonxoff = False, rtscts = False, dsrdtr = False), starting monitor "
```
Meapack activated on USB:
```
"Performing autodetection with 8 port / baudrate candidates: / dev / ttyUSB0 @ 500000, / dev / ttyUSB0 @ 115200, / dev / ttyUSB0 @ 250000, / dev / ttyUSB0 @ 230400, / dev / ttyUSB0 @ 57600, / dev / ttyUSB0 @ 38400, / dev / ttyUSB0 @ 19200, / dev / ttyUSB0 @ 9600
Trying port / dev / ttyUSB0, baudrate 500000
Handshake attempt # 1 with timeout 2.0s
Connected to: PackingSerial <id = 0x6bf6e7f0, open = True> (port = '/ dev / ttyUSB0', baudrate = 500000, bytesize = 8, parity = 'N', stopbits = 1, timeout = 2.0, xonxoff = False, rtscts = False, dsrdtr = False), starting monitor "
```
but the connection isn't stable and it's rejected.

I also tried with 250K, 115K and through USB: still the same problem. My Marlin 2.0.x is up to date with the bump distribution of 20210201 with Buffer 32, MEATPACK and BAUD_RATE_GCODE enable.

Regards,

tubalainen, 2021-02-14:
I just compiled the latest marlin bugfix 2.0.x and enabled meatpack.
Ive got the same issue as @ixendrax. I am also running the Ender 3 V2.
All runs fine until the plugin in octoprint is activated.

X-Ryl669, 2021-02-15:
@scottmudge I'm wondering if the meatpack decoding stuff shouldn't be better included as a Serial class so it's transparent to the gcode parser, unlike how it is now ?
My idea is like this pseudo code (not tested, but you'll get the idea):
```
template <typename SerialT>
struct MeatpackSerial : public SerialBase <MeatpackSerial < SerialT >> {
     typedef SerialBase< MeatPackSerial<SerialT> > BaseClassT;

  SerialT & out;
  uint8_t serialBuffer[2];
  uint8_t charCount;
  NO_INLINE size_t write(uint8_t c) { return out.write(c); }
  void flush()            { out.flush();  }
  void begin(long br){ out.begin(br); }
  void end()              { out.end(); }

  void msgDone() { out.msgDone(); }
  // Existing instances implement Arduino's operator bool, so use that if it's available
  bool connected()              { return Private::HasMember_connected<SerialT>::value ? CALL_IF_EXISTS(bool, &out, connected) : (bool)out; }
  void flushTX()                { CALL_IF_EXISTS(void, &out, flushTX); }

  bool available(uint8_t index) { 
    return charCount > 0 || (index == 0 && out.available()); 
  }
  int readImpl() {
   if (charCount > 0) return serialBuffer[charCount--];
   int r = out.read();
   if (r == -1) return r;
   meatpack.handle_rx_char((uint8_t)c);
   charCount = meatpack.get_result_char(serialBuffer);  
   return serialBuffer[charCount--]; 
  }

  int read(uint8_t index)       { if (index) return -1; return readImpl(); }
  bool available()              { return charCount > 0 || out.available(); }
  int read()                    { return readImpl(); }

  MeatPackSerial(const bool e, SerialT & out) : BaseClassT(e), out(out) {}
};

```

That should be completely transparent for Marlin's parser what's going on underneath (meatpack or not).

X-Ryl669, 2021-02-15:
If you can test this, and confirm that it's working, I'll modify the GCode parser/queue to remove all Meatpack specific stuff from there and include such serial class in the tree.

X-Ryl669, 2021-02-17:
Please try [this version](https://github.com/MarlinFirmware/Marlin/pull/21122), maybe it's working now

benlye, 2021-02-22:
Same problem here - can't connect with my BTT SKR 1.4 Turbo board if the Meatpack plugin is enabled.  Latest bugfix-2.0.x Marlin, latest Meatpack plugin (1.5.21), up-to-date version of OctoPrint (1.5.3).

```
2021-02-21 22:32:59,584 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2021-02-21 22:32:59,638 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
2021-02-21 22:32:59,638 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 115200
2021-02-21 22:32:59,643 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
2021-02-21 22:32:59,644 - octoprint.plugins.meatpack - INFO - [Serial]: Cannot query packing state -- port not open.
2021-02-21 22:32:59,650 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-02-21 22:32:59,654 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-02-21 22:33:02,472 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-02-21 22:33:02,476 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-02-21 22:33:06,474 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-02-21 22:33:06,478 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-02-21 22:33:10,474 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2021-02-21 22:33:10,497 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2021-02-21 22:33:10,557 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-02-21 22:33:10,827 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_autodetect, payload: {'commerror_text': 'No more candidates to test, and no working port/baudrate combination detected.'}
```

delroady, 2021-02-23:
Haven't tried the stuff from Feb 17, but I came here after having the same issue connecting with the Octoprint plugin enabled.  Figured I'd drop off what was showing up in terminal in case it could be of any use to someone.  (Just compiled this a day or so ago.  My first time doing the firmware and I just figured I put something in wrong)

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: PackingSerial<id=0xab403c30, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Trying port /dev/ttyACM0, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #2 with timeout 2.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 2.0s
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Connection closed, closing down monitor

jeff490, 2021-03-01:
Hello, Any progress on a fix?
Ender 3 pro with new 4.2.7 board
Marlin Bugfix as of today.
Wont connect to printer while meatpack plugin is enabled. Disable and it connects fine.

[33 more replies]