scottmudge/OctoPrint-MeatPack #14: Error Marlin\src\feature\meatpack.cpp:48:1: error: stray '@' in program compiling Marlin2

davbcr, 2021-02-10:
I've tried to compile adding a 
#define MEATPACK
command line into config.h and adding meatpack.h and meatpack.cpp into the src/features folder but i get this error.
It seems that PlatformIO doen't like @ 
maybe it's a stupid question but i don't know how to fix it (attention newbie onlne!)
meatpack.h and .cpp was merely capied and paste into new files into the folder mentioned above.

davbcr, 2021-02-10:
Simply i make copy and paste in the wrong way. 
But now i get some other errors:
1) In file included from Marlin\src\feature\meatpack.cpp:40:
Marlin\src\feature\meatpack.h:104:53: error: 'serial_index_t' does not name a type; did you mean 'serial_delay'?
  104 |   static void handle_rx_char(const uint8_t c, const serial_index_t serial_ind);
2) Marlin\src\feature\meatpack.cpp:187:10: error: 'MPCommand_TogglePacking' was not declared in this scope; did you mean 'MPCommand_EnablePacking'?
  187 |     case MPCommand_TogglePacking:   TBI(state, MPConfig_Bit_Active);   DEBUG_ECHOLNPGM("[MPDBG] TGL REC");   break;
      |          ^~~~~~~~~~~~~~~~~~~~~~~
      |          MPCommand_EnablePacking
Marlin\src\feature\meatpack.cpp:196:5: error: duplicate case value
  196 |     case MPCommand_EnableNoSpaces: { SBI(state, MPConfig_Bit_NoSpaces); DEBUG_ECHOLNPGM("[MPDBG] ENA NSP");
      |     ^~~~
3) Marlin\src\feature\meatpack.cpp:199:5: error: duplicate case value
  199 |     case MPCommand_DisableNoSpaces: { CBI(state, MPConfig_Bit_NoSpaces); DEBUG_ECHOLNPGM("[MPDBG] DIS NSP");
      |     ^~~~
4) Marlin\src\feature\meatpack.cpp:202:5: error: multiple default labels in one switch
  202 |     default: DEBUG_ECHOLNPGM("[MPDBG] UNK CMD REC");
      |     ^~~~~~~

compilation terminated due to -fmax-errors=5.

scottmudge, 2021-09-24:
Thanks for the report. As you may have found out, I do not maintain the Marlin implementation. This repository is primarily for the OctoPrint-side of things. If you still need help with the Marlin-side, please write an issue report on the Marlin repository.

closed 2021-09-24 by scottmudge