scottmudge/OctoPrint-MeatPack #21: Javascript/console errors

LazeMSS, 2021-09-24:
I see a lot of these:
```
packed_core.js?2d5b3bfb:1186 Error calling onStartup on view model MeatPackViewModel : TypeError: Cannot set properties of null (setting 'innerHTML')
    at MeatPackViewModel.self.updateAllText (http://localhost:5000/static/webassets/packed_plugins.js?34f3337b:6779:88)
    at MeatPackViewModel.self.onStartup.self.onUserLoggedIn (http://localhost:5000/static/webassets/packed_plugins.js?34f3337b:6765:26)
    at callViewModelIf (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1185:126)
    at http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1179:138
    at Pn (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:30:530)
    at Function.<anonymous> (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:51:66)
    at callViewModelsIf (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1179:95)
    at callViewModels (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1178:56)
    at fetchSettings (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1252:282)
    at sentryWrapped (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:630:45619)
```
It's this line: https://github.com/scottmudge/OctoPrint-MeatPack/blob/master/OctoPrint_MeatPack/static/js/meatpack.js#L150
I think you need to move the updates of the UI into the onAllBound

```
packed_core.js?2d5b3bfb:1248 Could not bind view model MeatPackViewModel to target #meatpack_total_tx_string : Error: You cannot apply bindings multiple times to the same element.
    at p (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:131:3221)
    at k (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:131:2747)
    at Object.a.vc (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:132:3231)
    at http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1247:8
    at Pn (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:30:530)
    at Function.<anonymous> (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:51:66)
    at http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1242:26
    at Pn (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:30:530)
    at Function.<anonymous> (http://localhost:5000/static/webassets/packed_libs.js?1fe9f29c:51:66)
    at bindViewModels (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:1233:106)
    at sentryWrapped (http://localhost:5000/static/webassets/packed_core.js?2d5b3bfb:630:45619)
```
Not sure about this one

scottmudge, 2021-09-24:
Interesting, I'll take a look at it. Thank you for the report.

LazeMSS, 2021-09-28:
@scottmudge  not sure if you get notifications on PR - anyways: https://github.com/scottmudge/OctoPrint-MeatPack/pull/22
Take a look and let me know what you think. I was unable to test it because my one printer died but there is nothing touched in the way data is handled.