scottmudge/OctoPrint-MeatPack #29: VIRTUAL printer broken after installing MeatPack plugin

arekm, 2023-04-18:
MeatPack overrides serial port opening (could it not do that?) which causes Virtual Printer (available by default in any OctoPrint installation) to be broken:

````
2023-04-18 07:57:53,271 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port VIRTUAL, baudrate 115200 from hook meatpack: SerialException: '[Errno 2] could not open port VIRTUAL: [Errno 2] No such file or directory: 'VIRTUAL'' @ comm.py:_open_serial:3916
Traceback (most recent call last):
  File "..../OctoPrint/venv/share/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: 'VIRTUAL'
````

arekm, 2023-04-18:
What these numbers refer to? Issues somewhere but where?

```
        needs_parity_workaround = get_os() == "linux" and os.path.exists("/etc/debian_version")  # See #673
````

and

````
        # Set close_exec flag on serial handle, see #3212
````

scottmudge, 2023-04-18:
I'm confused about why you want to have MeatPack enabled when using a Virtual printer? 

It needs to hook into and override the serial port creation so it can interface directly with the printer and send it the compressed data. There is no callback or event hook for OctoPrint which allows outgoing serial data to be arbitrarily changed before it goes to the printer. At least with the performance and granularity required here.

--

Those numbers refer to issues on the original OctoPrint repo. The serial code is mostly copied from what is already used internally.

arekm, 2023-04-18:
Wanted to implement meat pack in virtual printer (virtual printer is used for testing things in octoprint without real hardware). Thanks for pointers.

scottmudge, 2023-04-18:
Doing so would require implementing the associated unpacking/decompression code on the virtual-printer side. I do know that Prusa's MK404 printer emulator w/ the MeatPack PR applied to the firmware image works with OctoPrint + MP. In fact, this is how they tested the PR before approving it.